mantidproject/mantid #18990: Mantid crashes when checking the sample log of some workspaces

AntonPiccardoSelg, 2017-02-24:
Querying the sample log of the system test data `SANS2D00005512.nxs` creates an unhandled exception. Strangely this is only true for the child workspaces 1 to 11. Querying the sample log from child workspaces 12 and 13 works as expeceted.

### Expected behavior

The sample logs can be queried on all child workspaces of a group workspace;

### Actual behavior

Mantid crashes when querying the sample logs on some child workspaces.

### Steps to reproduce the behavior

See above

### Platforms affected

Seen on Ubuntu 16.04

AntonPiccardoSelg, 2017-02-24:
Initial scoping of the issue. The issue occurs for the `selog>Tank_Pressure` entry. The point fo failure is
[here](https://github.com/mantidproject/mantid/find/master#L2023:L2024).

martyngigg, 2017-02-24:
This is most likely a filtering issue.

When you say querying, can I just extract the log in python?

AntonPiccardoSelg, 2017-02-24:
I stumbled across it via the GUI. Right-click on the workspace, then select `Sample Logs...`. This will cause the crash.

martyngigg, 2017-02-24:
It looks like the problem here is that log has 1 value in it and then filtering has removed that value and we are left with an empty log. I will have to investigate further as to whether the filtering has been applied appropriately.

martyngigg, 2017-03-06:
@wdzhou Would you have time to look into this issue? You understand the log filtering better than I do.

martyngigg, 2017-03-09:
[image] [Filtering Crash](https://trello.com/c/xdbSs4wJ/312-filtering-crash)

Pasarus, 2018-07-06:
I've found a similar bug which throws an unexpected exception, this happens when you attempt to show sample logs for a group workspace. This can be done by highlighting the group workspace and then using the option workspace->show logs in MantidPlot as it isn't disabled here like in the right click menu.

martyngigg, 2019-07-15:
This appears to have been fixed in the current version.

closed 2019-07-15 by martyngigg