mantidproject/mantid #19005: Ability to select a workspace in the ADS as the transmission run

raquelalvarezbanos, 2017-02-27:
This is something scientists would like to have.

In the new reflectometry GUI, if you have a workspace loaded in the ADS called for example `TOF_13460`, you can enter `TOF_13460` in the table, and the interface will pick the run from the ADS instead of loading into Mantid. This also works for transmission runs, so the table, will be able to reduce the following:

[image]

just by retrieving the workspaces from the ADS. However, this is not possible if you specify the transmission run from the `Settings` tab. This issue is to add this functionality.

PranavBahuguna, 2017-03-15:
I think the PR for #18707 might automatically solve this issue since it allows the transmission runs in the `Settings` tab to be used in preprocessing (where runs can be obtained from ADS directly instead of loading them) without reuse in processing.

PranavBahuguna, 2017-04-26:
PR #19111 fixes this issue.

closed 2017-04-26 by PranavBahuguna