mantidproject/mantid #19012: Implement slicing by log value

raquelalvarezbanos, 2017-02-28:
Design sent by Max:

[image]

I believe we could use `FilterByLogValue`, for which we would have to expose properties `MinimumValue` and `MaximumValue`. The first step before implementing this in the interface is to understand how the algorithm works (for instance, what happens if two non-consecutive blocks of data are in the allowed range of log value?). The second step is to write a script similar to the script originally sent by Max (can be found [here](https://github.com/mantidproject/mantid/files/639344/EventRef.zip)), discuss it with scientists and implement the functionality in the interface once they are happy with it.

PranavBahuguna, 2017-03-15:
Currently blocked by #18500

PranavBahuguna, 2017-03-24:
After discussion with instrument scientists, there are now new details on how to implement slicing by log value:
- In addition to the log value type selection box, there will be a text box for entering a number of comma separated values in a manner similar to custom slicing.
- We are to apply the algorithm `FilterByLogValue` on each pair of values in the text box by passing said values as `MinimumValue` and `MaximumValue` respectively. For example if we have the following values in the text box `0, 10, 20, 30`, then we will perform `FilterByLogValue` on `0, 10`, `10, 20` and `20, 30` respectively. We will pass the value from the selection box to `LogName`.

closed 2017-05-22 by quantumsteve