mantidproject/mantid #19020: Default facility and instrument setting issue

gvardany, 2017-02-28:
When altering the facility through config service, the instrument stays the same. Similarly when altering the instrument, the facility stays the same.
This seems counterintuitive and error prone.
There are few places at least, where this causes a problem, since one can get a different result when asking the facility from instrument, or from config service directly.
Of course these cases (e.g. file prefix in MultiFileNameParser) can be cured, but it is probably better to change the `ConfigService` itself, to change both in sync.

```
print config['default.facility']
print config['default.instrument']
config.setFacility('ILL')
print config['default.facility']
print config['default.instrument']
print config.getInstrument().facility().name()
```

gvardany, 2017-02-28:
@martyngigg Do you think this is reasonable?

martyngigg, 2017-03-01:
@gvardany You're right this seem confusing. What do you propose? Should we just clear the `default.instrument` when setting the facility?

gvardany, 2017-03-01:
That's one possibility, do you think the 'empty' instrument will not cause problems? Alternatively, one could do similar to what first time setup GUI does, i.e. when changing the facility it sets the instrument to the first (alphabetically) instrument belonging to it.

martyngigg, 2017-03-01:
On very first setup both default strings are actually empty so it shouldn't cause problems. First in the list is also a reasonable approach though I think.

gvardany, 2018-11-27:
This is not particularly urgent.

closed 2018-11-27 by gvardany