mantidproject/mantid #19038: Crash opening Live Data dialog

gemmaguest, 2017-03-02:
### Expected behavior
Mantid should not crash

### Actual behavior
Mantid crashed when opening the Live Data dialog

### Steps to reproduce the behavior
Unfortunately I have not been able to reproduce this. Here is what the user did at the time of the crash:
- The user was running through the examples here: http://www.mantidproject.org/MBC_Live_Data_Simple_Examples#Do_it_yourself_examples
- The user got as far as clicking Load -> Live Data at this step: http://www.mantidproject.org/MBC_Live_Data_Simple_Examples#Starting_a_live_data_session
- The Live Data dialog started up, but mantid then crashed before the user had interacted with it.

### Platforms affected
This was seen on Windows using Mantid 3.9.0

NickDraper, 2017-06-29:
Leaving open in case we see this again, but unassigning until then

NickDraper, 2017-06-29:
Not seen in 3.10 release testing, closing

closed 2017-06-29 by NickDraper