mantidproject/mantid #19040: MonitorLiveData algorithm hangs/crashes on cancel

gemmaguest, 2017-03-02:
### Expected behavior
The algorithm should not hang and Mantid should not crash.

### Actual behavior
The `MonitorLiveData ` algorithm sometimes hangs and Mantid sometimes crashes.

### Steps to reproduce the behavior
The problem is intermittent, but I have seen it several times following this procedure:
- Open the "Help" menu and select "First Time Setup"
- Change the default facility to TEST_LIVE and Default instrument to ISIS_Event and click "Set"
- Run FakeISISEventDAE with the default params
- Click Load -> Live Data
- Set Instrument to ISIS_Event and Connection to event.
- Set the output workspace name to "live" and tick "Preserve Events"
- Click Run.
- Click on the Details button to open the Algorithm progress dialog.
- Click Cancel next to the the FakeISISEventDAE algorithm. It should disappear.
- Click Cancel next to the MonitorLiveData algorithm. It changes to "Canceling" but never finishes.
- After a short time, or on repeating the process, mantid sometimes then crashes.

### Platforms affected
Seen on Windows 7 in v3.9.0 and in master at commit 2d86764ed4

gemmaguest, 2017-10-16:
We haven't been able to reproduce the original problem in v3.11.0 yet but I've seen another crash related to cancelling the MonitorLiveData algorithm. I had run the above instructions, restarted FakeISISEventDAE, and started a new MonitorLiveData algorithm (so I now had 3 algorithms running). I cancelled all 3, but the second MonitorLiveData algorithm did not finish, and after a short while Mantid crashed. (I had actually just opened the Load dialog too at that point but I doubt that's related).

gemmaguest, 2017-10-30:
I've managed to reproduce the original problem in 3.11.20171024.2048 on Windows 7. It happened in 2 out of 2 attempts on the second run through the instructions and when cancelling both algorithms in quick succession (FakeISISEventDAE first). Something I noticed at least once was that cancelling MonitorLiveData didn't seem to do anything when I first hit Cancel (the button almost immediately reverted back to saying Cancel), so I clicked Cancel again and that's when it hung saying Cancelling before crashing a short while later.

NickDraper, 2020-01-31:
Should be fixed by #27077

closed 2020-01-31 by NickDraper