mantidproject/mantid #19068: Script Mantid Python Window doesn't expand user variables in Linux

DimitarTasev, 2017-03-06:
Quick sample code that will fail to find the workspace because it's written as:
`'~/Documents/IMAT00003680.nxs'
if changed to the full path:
`'/home/<USER>/Documents/IMAT00003680'` 
it passes successfully.
The issue could be in the algorithms, because they are the ones responsible for expanding the path (I think)
```
idf_path = r'~/Documents/mantid/instrument/IMAT_Definition.xml'
# The path of the grouping file
grouping_file_path = r'~/Documents/imat_2bank.cal'

# After focusing the workspace will be cropped so only data between these TOF values are shown
# this should rectify issues with scale
focused_tof_xmin = 2000
focused_tof_xmax = 19995

# ---- Script ------- # 
# Generate the name we will be using for all workspaces 
out_name_generic = 'IMAT0000123331'

import os
run_path = '~/Documents/IMAT00003680.nxs'
# Generate path the input data is located at
ws = Load(Filename=run_path, OutputWorkspace=out_name_generic)
ws = NormaliseByCurrent(
    InputWorkspace=ws, OutputWorkspace=out_name_generic + '_norm')

# Replace the IDF with the new version
LoadInstrument(
    Workspace=ws,
    Filename=idf_path,
    MonitorList='0,1,2,3,4,5,6,7,8,9,10,11,12,13',
    RewriteSpectraMap='True')

# Focus
ws = ConvertUnits(
    InputWorkspace=ws,
    OutputWorkspace=out_name_generic + '_d',
    Target='dSpacing')
ws = DiffractionFocussing(
    InputWorkspace=ws,
    OutputWorkspace=out_name_generic + '_focused',
    GroupingFileName=grouping_file_path)

# Convert focused WS back to TOF
ws = ConvertUnits(
    InputWorkspace=ws,
    OutputWorkspace=out_name_generic + '_focused_tof',
    Target='TOF')
# Crop to valid range
ws = CropWorkspace(
    InputWorkspace=ws,
    OutputWorkspace=out_name_generic + '_focused_TOF_crop',
    XMin=focused_tof_xmin,
    XMax=focused_tof_xmax)
```

DTasev, 2017-06-30:
I have removed the Bug label, but I am not sure what label to put for a missing feature.

As for the expansion, I agree it should follow `os.path.expanduser`, as this is what I expected would happen.

martyngigg, 2017-06-30:
I think it makes most sense for this behaviour to be added in the `FileProperty` using the rules laid down in https://docs.python.org/2/library/os.path.html#os.path.expanduser. This is then nice and easy to unit test too.

PR #20007 (2017-08-09): Added expanduser to Load in simpleapi
Description of work.

Added expansion of user's home directory (denoted by `~`) to `FileProperty`, so now files with `~` in the path can be loaded with standard behaviour, whereas previously they were not found.

**To test:**

Place a .nxs file in your home directory (~), or some subdirectory. From the script interpreter in MantidPlot, run `Load(Filename="~/<path-to-file>", OutputWorkspace="ws")`. The file should be loaded as expected.

Fixes #19068 . 

**Release Notes** 
[here](https://github.com/mantidproject/mantid/pull/20007/files#diff-d43ebaec89d7bc36 [... cut]
closed 2017-08-09 by martyngigg