mantidproject/mantid #19071: Clarify documentation for LoadInstrument

DavidFair, 2017-03-06:
The `RewriteSpectraMap` option does within `LoadInstrument` is not very clear for example what happens if it is set to false. 
Update and clarify the documentation with the help of @DimitarTasev

DavidFair, 2017-03-10:
Actually looking at the parameter description it describes the behaviour pretty well - I can't see any reason to change it. What do you think @DimitarTasev ?

DimitarTasev, 2017-03-13:
@DavidFair We could clarify what happens if set to `unset` and `false`, as this makes me think why did it work when I set it to `false`, did it keep the current structure in the nexus file (likely), and assign the matching bank IDs from the IDF to the matching spectra numbers? What is the difference between `unset` and `false`? 

We could extend the documentation to clarify more about the functionality.

PR #20499 (2017-09-20): Improve documentation for RewriteSpectraMap property of LoadInstrument
Description of work.

**To test:**

- Review documentation for `LoadInstrument` and see that the description of the `RewriteSpectraMap` is correct and easy to understand

Fixes #19071. 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and  [... cut]
closed 2017-09-20 by martyngigg