mantidproject/mantid #19130: WorkspacePropertyWithIndex property

LamarMoore, 2017-03-14:
This issue is with regards to the new indexing features being implemented in Mantid.

This property will allow indexing translation and validation to occur as part of the declaration of the property. Users would declare the property using something like:
```cpp
declareProperty(make_unique<WorkspacePropertyWithIndex>("InputWorkspace", "", 
                          Direction::Input, UseSpectrumNumber|UseDetectorID|UseWorkspaceIndex), 
                          "Input workspace and selected indices loaded simultaneously.");
```
Users would then be able to retrieve indexes in the following manner:

```cpp
//c++11
const MatrixWorkspace_const_sptr inputWS;
  const IndexSet indexSet;
  std::tie(inputWS, indexSet) = getProperty("InputWorkspaceWithIndex");

//c++17 
auto [inputWS, indexSet] = getProperty("InputWorkspaceWithIndex");
```
```python
# python
inputWs, indexSet = getProperty(InputWorkspaceWithIndex);
```
N.B This issue relates to Mantid 4.0 although it could be added to before that.

jmborr, 2017-05-04:
@LamarMoore   so what would be the syntax for passing a workspace name and an index range to this property?

LamarMoore, 2017-05-04:
@jmborr You will pass the workspace name when declaring the property and specify what index types are accepted. The GUI will create something similar to the input for an array property which will capture the indices. The indices would then be accessed as outlined in the description above.

jmborr, 2017-05-04:
I still don't understand. Can you give a specific use case?

LamarMoore, 2017-05-04:
If you look at the `ChangePulseTime` algorithm it takes a workspace and an optional list of indices for applying the algorithm. With the new property, the two properties would be combined so users could declare a workspace and at the same time specify that they want to capture workspace indices. The GUI will present the users with a text box where they can enter a list or a range of indices. The indices will be translated by the new `IndexInfo` which has been added to MatrixWorkspace. Indices will also be immediately validated so the users will be forced to enter correct information. The information will be input in the same way as it is done for `ArrayProperty`. Is the clearer?

jmborr, 2017-05-04:
@LamarMoore Yes, this example is crystal clear, thanks :bowtie:

PR #19940 (2017-08-17): IndexProperty Implementation Prototype
**DO NOT MERGE**
**NOT READY FOR REVIEW**
This is an alternative implementation to the implementation of an IndexProperty see [here](https://github.com/mantidproject/mantid/pull/19699) for the first attempt. This [design](https://github.com/mantidproject/documents/pull/42) outlines the rational behind developing this property type. 

**To test:**
Code review. Ensure the property displays correctly in the GUI and can be accessed from python correctly. The `ChangePulsetime2` algorithm was implemented specifically to demonstrate the use of this property.

Fi [... cut]
closed 2017-08-17 by martyngigg