mantidproject/mantid #19135: Masking issues for certain SANS files

AntonPiccardoSelg, 2017-03-14:
Diego reported that altering some of the mask files on a pixel level did not have any effect. 
1. Look if this can be seen/reproduced directly in the instrument viewer
2. Check if this is caused on the ISIS reduction side

AntonPiccardoSelg, 2017-03-15:
Investigations showed that when the masking display is being used through the Mantid ISIS SANS GUI, masked areas are set to negative values. They cause an issue when being displayed in a log scale in the instrument view. The values and the masking is correct in principle and it is ok that negative values are not handled by the instrument view.

In principle we should leave masked values masked. It is not clear why this had been implemented like this the first time round, but it should be simple enough to change. This requires discussion with the instrument scientists. 

Since the issue is only cosmetic, the priority level is being decreased.

smk78, 2020-07-16:
Should there be a separate mask array?

RichardHeenan, 2020-07-27:
Need to check current behaviour. Masked pixels ought to display in a "masked colour" regardless of whether log or linear intensity is chosen.

closed 2021-09-06 by stale[bot]