mantidproject/mantid #19226: Implement different processing instructions for transmission runs

raquelalvarezbanos, 2017-03-23:
From Jos:

> Ideally, what would happen would be that there are different processing instructions for the direct beam as from the reflected beam. This is because the direct beam droops under gravity and so we set a large range in order to catch all of the neutrons. The reflected beam has the gravity somewhat cancelled because it bounced from the surface (though at some point we will come to try correcting this), so generally we don’t want as large a range.
I think that actually it might be possible to put the multidetector transmission processing instructions into the parameter file, since they really don’t change much. Max, your thoughts/objections on this?
Anyway, I think that as a general principle, we want to integrate over the entire direct beam for the TRANS, and select what we integrate over for the reflected beam. I don’t think that we mind too much exactly how it is executed.

raquelalvarezbanos, 2017-03-23:
The question to ask scientists is if we need a second `ProcessingInstructions` property in `ReflectometryReductionOneAuto`. Then `ReflectometryReductionOneAuto` can look into the parameter file for the value of this property and pass it on to `ReflectometryReductionOne` as it does with other properties such as `WavelengthMin` etc.

raquelalvarezbanos, 2017-03-27:
As a temporary fix I will fix processing instructions of the transmission run in https://github.com/mantidproject/mantid/issues/19252.

Pasarus, 2018-09-18:
Some of this is implemented in: #23431

gemmaguest, 2019-07-24:
This was done as part of #14000

closed 2019-07-24 by gemmaguest