mantidproject/mantid #19252: Fix processing instructions applied to transmission run

raquelalvarezbanos, 2017-03-27:
`ReflectometryReductionOneAuto` is not applying `ProcessingInstructions` properly to the transmission run.

### Expected behavior

If `ProcessingInstructions` is given as a range of pixels, the whole range should be used for the transmission run.

### Actual behavior

Only the first pixel is applied to the transmission run.

### Steps to reproduce the behavior

Run the following script:

```
LoadISISNexus(Filename=r'\\isis\inst$\ndxoffspec\instrument\data\cycle_16_3\OFFSPEC00039898.nxs', OutputWorkspace='TOF_39898')
LoadISISNexus(Filename=r'\\isis\inst$\ndxoffspec\instrument\data\cycle_16_3\OFFSPEC00039904.nxs', OutputWorkspace='TRANS_39904')
ReflectometryReductionOneAuto(InputWorkspace='TOF_39898',
                                            AnalysisMode='MultiDetectorAnalysis',
                                            ProcessingInstructions='390-420',
                                            ThetaIn=1,
                                            FirstTransmissionRun='TRANS_39904',
                                            MomentumTransferStep=0.01,
                                            OutputWorkspaceBinned='IvsQ_binned_39898',
                                            OutputWorkspace='IvsQ_39898',
                                            OutputWorkspaceWavelength='IvsLam_39898')
```

Check the history of workspace `IvsQ_binned_39898`, `ProcessingInstructions` was `390` instead of `390-420`:

[image]

### Platforms affected

All.

raquelalvarezbanos, 2017-03-27:
Note that there's a workaround at the moment: you can set `StrictSpectrumChecking` to false and processing instructions are passed correctly on to `CreateTransmissionWorkspaceAuto`.

raquelalvarezbanos, 2017-03-27:
The problem is [this](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/ReflectometryReductionOne2.cpp#L257-L265) block of code, which was copied from version 1 of the algorithm (see [here](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/ReflectometryReductionOne.cpp#L730-L740)):

If “StrictSpectrumChecking” is true, which is by default, the algorithm gets the spectrum map of the detector workspace –which only has one element mapping spectrum number 391 to workspace index 0, as the detector workspace has been already summed in wavelength – and uses it to create new processing instructions. Because the spectrum map only has one one element {391 : 0}, the new ProcessingInstructions that are passed to CreateTransmissionWorkspaceAuto are 390. I am not sure why the code was doing this. The fix is to remove it, but I am can't be sure that it won't break other functionality I am not aware of.

gemmaguest, 2017-03-30:
I have made an alternative fix, which fixes the logic in [this](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/ReflectometryReductionOne2.cpp#L257-L265) code rather than removing it. This was easier to do as part of the work for #18829 so it will be part of the PR for that work.

I have not been able to add a unit test yet because the test case hits the separate bug #15749 .

PR #19502 (2017-05-22): Add option to `ReflectometryReductionOne` to sum in Q
This PR adds an option, `SummationType`, to `ReflectometryReductionOne` to allow summation in Q, as well as in the default lambda. The option is also available in `ReflectometryReductionOneAuto` and via the Options field in the reflectometry GUI. In addition, there is an option, `ReductionType`, to specify what type of reduction should be done when summing in Q (`DivergentBeam` or `NonFlatSample`).

Documentation will follow in a separate PR.

**To test:**

Interim test scripts are below, but the results still require scientific validation. [... cut]
closed 2017-05-22 by martyngigg