mantidproject/mantid #19256: Global fitting via scripts is broken

jmborr, 2017-03-27:
Pull request [#18362](https://github.com/mantidproject/mantid/pull/18362)  has broken the global fitting scripts in the [script repository](https://github.com/mantidproject/scriptrepository/tree/master/indirect_inelastic/models).

Benchmark data  and script to fix this:  
[fit_irs26176_with_Teixeira.py](https://github.com/mantidproject/scriptrepository/blob/master/indirect_inelastic/models/TeixeiraWaterSQE/v3.9/fit_irs26176_with_Teixeira.py)  
[irs26173_graphite_res.nxs](https://github.com/mantidproject/scriptrepository/blob/master/indirect_inelastic/models/TeixeiraWaterSQE/irs26173_graphite_res.nxs)  
[irs26176_graphite002_red.nxs](https://github.com/mantidproject/scriptrepository/blob/master/indirect_inelastic/models/TeixeiraWaterSQE/irs26176_graphite002_red.nxs)

jmborr, 2017-03-27:
@mantid-roman I'm trying to find out what went wrong, but I'd appreciate it if you could give me a hand.

mantid-roman, 2017-03-27:
I'll have a look

jmborr, 2017-03-27:
thanks a lot!

mantid-roman, 2017-03-27:
@jmborr Could you explain what is broken?

mantid-roman, 2017-03-27:
I think I know what happened. Sorry it was my fault. The fitting parameters should be OK but the computed spectra are wrong. The fix is in the patch release and in `master`.

jmborr, 2017-03-27:
wow, that was fast. Can you please tell me the sha1 ID for the commit to master so that I can learn what needed to be fixed?

mantid-roman, 2017-03-27:
This is the pull request: https://github.com/mantidproject/mantid/pull/19161

jmborr, 2017-03-27:
I just compiled master, they're working fine now :smile:

jmborr, 2017-03-27:
Fixed with #19161

closed 2017-03-27 by jmborr