mantidproject/mantid #19258: BASIS definition broken/inefficient

OwenArnold, 2017-03-28:
This issue has been discovered after I added sanity checks for #19015

For the basis definition 20160 (8 banks x 2520) detectors have conflicting IDs. This means that only 4 banks for the BASIS definition are actually loaded out of the possible 8. 

This is at least a serious performance problem. Since the full Detector objects are still created an added to the instrument. That leaves **10080** effectively orphaned detectors attached to the instrument. The only looks normal because the duplicate ids to not get added to the detectorid list [see here](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument.cpp#L701-L704). It would be better IMHO if the BASIS definition only contained detectors that were actually supplied with a valid ID.

jmborr, 2017-03-28:
@OwenArnold I'm looking at the current BASIS definition file, `BASIS_Definition_20140101-.xml`. There are no duplicate pixels as far as I can tell, but maybe the way the pixels are declared induces  duplication. Below is the scheme for listing the pixels for one of the banks

```
<!-- Listing the pixels for bank 3 -->

<component idlist="bank3" type="bank3">
  <location/>
</component>

<type name="bank3">
  <!-- Start listing the pixel details -->
  <component type="pixel">
    <location name="8197" x="-0.487260681056" y="0.21631015625" z="-2.39457769788">
       <neutronic p="179.356" r="4.72838" t="161.199"/>
    </location>
      <parameter name="Efixed">
         <value val="2.08275"/>
      </parameter>
  </component>
  <!-- More pixels follow for bank 3-->
</type>

<idlist idname="bank3">
  <id val="8197"/>
   <!-- More ID's follow for bank 3--> 
</idlist>
```
There's an orphaned tag `<location/>` whithin the `<component>` section.

OwenArnold, 2017-03-29:
id lists for bank1, bank2, bank3 and bank4 are reused. for example ``idlist="bank1"``. These Appear in sub-banks of "silicon" as well as those banks directly attached to the top level instrument.

jmborr, 2017-03-29:
@OwenArnold  Ouch! my ignorance of IDF syntax is showing off :stuck_out_tongue_winking_eye:
Thanks a lot for finding the duplicity.

PR #19273 (2017-04-05): BASIS definition broken/inefficient
**To test:** Donwload [BSS_66008_sqw.nxs.zip](https://github.com/mantidproject/mantid/files/883445/BSS_66008_sqw.nxs.zip), unzip, and run the following script to verify reduction has not been broken
```
BASISReduction(RunNumbers="66008", ReflectionType="silicon111", EnergyBins=[-120,0.4,120], MomentumTransferBins=[0.3, 0.2, 1.9])
old = LoadNexus(Filename="BSS_66008_sqw.nxs")
ratio = (mtd["BSS_66008_sqw"]-old)/old
plotSpectrum(ratio, range(ratio.getNumberHistograms()))
```
Values in `ratio` should not be bigger than ~10^(-5)

Fixes #19258.

[release notes](ht [... cut]
closed 2017-04-05 by martyngigg