mantidproject/mantid #19281: Fitting interfaces stopped handling ties to a constant when loading a function from the clipboard

jmborr, 2017-03-31:
Example: The fit function string `name=LinearBackground,A0=0,A1=0,ties=(A0=0)`:

1. Produces a function without the tie when loaded into interface **Fit Function**. One can later add the tie manually by invoking **Fix** from  parameter A0 pull-down menu
2. Produces a function without the tie when loaded into interface **Multi dataset fitting**. Trying to manually fix parameter A0 in the interface results if segfault and shuts down Mantid with the following error:
```
ASSERT failure in QVector<T>::operator[]: "index out of range", file /usr/include/qt4/QtCore/qvector.h, line 359
Aborted (core dumped)
```

Tested on Ubuntu 1604 with **master** branch with build 3.9.20170331.1511

jmborr, 2017-03-31:
@mantid-roman Hi Roman, I suspect this has to do the recent change regarding the lack of a ParameterTie for a fixed parameter.

mantid-roman, 2017-03-31:
Yes. I'll have a look.

jmborr, 2017-04-06:
@mantid-roman Is this proves to be too much of a fuss, would you consider rolling back the changes and let parameter fixes instantiate parameter tie references? To me there's nothing conceptually wrong assigning a tie to a parameter.

jmborr, 2017-04-12:
looks good to me
:shipit:

PR #19321 (2017-04-13): Fit browser ties bug
Put in special code for fixed but not tied parameters.

**To test:**

Instructions are in the issue description.

Fixes #19281.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/methods/functions?
- [ ] Are there unit/system t [... cut]
closed 2017-04-13 by peterfpeterson