mantidproject/mantid #19289: Mask Detectors Fails with Invalid Unordered_map key

DavidFair, 2017-04-03:
### Expected behavior
Masking detectors with a spectrum list (e.g. `0-58`) should work
### Actual behavior
Algorithm fails with
```
Error in execution of algorithm MaskDetectors:
invalid unordered_map<K, T> key
```

### Steps to reproduce the behavior
Following script:
```
CreateSampleWorkspace(OutputWorkspace='test', Function='Flat background', NumBanks=1, XMax=2, BinWidth=1, BankPixelWidth=10)
MaskDetectors(Workspace="test", SpectraList=range(0, 51))
```

### Platforms affected
All

DavidFair, 2017-04-04:
This is caused by the line
`auto ws_index : WS->indexInfo().makeIndexSet(spectraList)`

Inside `IndexInfo` the line `m_spectrumNumberToPartition.at(spectrumNumber)` fails with the "Invalid unordered_map<K, T>" message.

DavidFair, 2017-04-04:
Tagging @SimonHeybrock for help with this issue.

SimonHeybrock, 2017-04-04:
@DavidFair I am on leave until next Tuesday (probably) so I cannot look at this immediately.

The cause of the issue might be an invalid spectrum number. Are you sure that `0` is a valid spectrum number for this workspace? If not, check the validation code in `MaskDetectors`.

SimonHeybrock, 2017-04-10:
I had a look at the old version of the code: It did silently ignore invalid spectrum numbers (0 in this case). I am not sure that is a sensible behavior? Maybe the thing to do is to have a better exception message?

DavidFair, 2017-04-19:
@SimonHeybrock 

I'm not sure if we were purely C++ throwing when we use index `0-n` makes perfect sense. 
However this occurs in Python as we do something like `range(num_banks + 1)` and `range(1, num_banks + 1)` doesn't look as clean to me.

We could have this algorithm to change the 0 to 1 and log something as a warning or notice?

Also tagging @AndreiSavici in with #19399 as this is a duplicate issue

SimonHeybrock, 2017-04-20:
@DavidFair This is nothing do to with C++ or Python conventions. The spectrum numbers are mainly for the user interface (and files). If you do not like that spectrum numbers (often) start at `1` you could use access via the workspace index.

Personally I believe that algorithms should always fail if a bad index is entered, since this might indicate that something is wrong. Warnings in the log window tend to be ignored, so while warning is better than silently ignoring, there is still a risk of having a buggy script, which may result in wrong results in publications.
Ultimately, it is not for me (alone) to decide, so probably this should to be discussed somewhere else.

DavidFair, 2017-07-17:
Issue was resolved in ISIS Powder and has to be resolved on a case-by-case basis on the caller side.

closed 2017-07-17 by DavidFair