mantidproject/mantid #19303: Improve ctest support with multi-configuration generators

quantumsteve, 2017-04-04:
### Expected behavior

CMake recently added `$<CONFIGURATION>` which can be used for setting ctest variables that depend on the build configuration (Debug, Release, etc.).

http://stackoverflow.com/questions/9467510/how-to-pass-ctest-configuration-type-using-add-test-in-cmake

https://cmake.org/gitweb?p=cmake.git;a=commitdiff;h=07934636

### Actual behavior

Mantid implements a workaround for python tests and MSVC. The added functionality could both make this much simpler and work with more build configurations like RelWithDebInfo.

https://github.com/mantidproject/mantid/blob/844e45473e35096d6a948fd7d573a285b19284db/buildconfig/CMake/FindPyUnitTest.cmake#L27

### Platforms affected

Windows, MacOS

quantumsteve, 2017-04-04:
I successfully used this variable to [set the PythonPath for SpinWaveGenie's test suite](https://github.com/SpinWaveGenie/SpinWaveGenie/blob/0b423e0b6c247a38c95d9fb28703ef918a9b0f10/pythonapi/test/CMakeLists.txt#L13)

`$<CONFIGURATION>` seems to give the same result for multi-configuration and single-configuration builds.

martyngigg, 2017-04-05:
This is good. I remember this being a pain when I first set did those tests.

PR #21086 (2017-11-03): Use generator expressions in pyunittest_add_test
Description of work.

Merges the handling of different ctest configurations into a single `add_test` command. At the same time we add support for Xcode too.

**To test:**

The builders should check that the execution behaviour has not been broken. Double check the python tests run and pass. A code review should be sufficient.

Fixes #19303

**Release Notes** 

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### C [... cut]
closed 2017-11-03 by SimonHeybrock