mantidproject/mantid #19318: Create mock ups for QuickRun GUI

martyngigg, 2017-04-07:
Create either a Balsamiq or Qt Designer mock up of the interface and tabs.

Discuss these with scientists to get general agreement that the design is okay.

martyngigg, 2017-04-07:
@louisemccann Please attach the mock ups to this issue so that we can keep track of them and have a history that they were done.

louisemccann, 2017-04-07:
[image]
[image]
[image]
[image]

Initial mockups in Balsamiq

martyngigg, 2017-04-07:
Thanks. Let's keep this issue up to date as we get feedback and update them so that we should converge on a design.

closed 2021-03-03 by stale[bot]