mantidproject/mantid #19339: Performance regression caused by changes to IndexInfo

SimonHeybrock, 2017-04-11:
See, e.g., `ConvertSpectrumAxis` and `DiffractionFocussing2` here: http://builds.mantidproject.org/view/All/job/master_performancetests2/Master_branch_performance_tests/.

OwenArnold, 2017-04-11:
I've taken a look at `ConvertSpectrumAxis` with a profiler. It looks like the majority of the time is being spent in `setInstrument` constructing `API::DetectorInfo` and `API::ComponentInfo`, (both taking the same amount of time).

OwenArnold, 2017-04-11:
Looking at `DiffractionFocussing2` it seems that `IndexInfo` is the offender.

SimonHeybrock, 2017-04-18:
@OwenArnold For `DiffractionFocussing2`, I notice that `LoadEmptyInstrument` is taking a very long time? It is outside the test cases so does not affect the performance test.

PR #19410 (2017-04-24): Fix performance regression caused by changes to IndexInfo
See title and issue. Probably this can be done in a nicer way, but since `IndexInfo` is still in early development I suspect there will be further changes in the near future so I kept things simple for now.

**To test:**

Code review.

Fixes #19339 (probably, though further improvements might be necessary at some point).

No release notes, issue introduced after last release.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [... cut]
closed 2017-04-24 by OwenArnold