mantidproject/mantid #19385: BinMD too slow for Sliceviewer (300 times slower than 3.9)

VickieLynch, 2017-04-14:
Reported from TOPAZ current experiment user:
### Expected behavior
For 8 Gb MDEvent file, BinMD takes 0.05 seconds for each slice rebinned with MantidPlot3.9
### Actual behavior
For the same file with nightly, BinMD takes 15 seconds for each slice which makes SliceViewer with auto rebin too slow to use.  Profiler shows the time is mostly in ExperimentInfo.cpp line 99 set/get instrument.
### Steps to reproduce the behavior
LoadMD with large MDEvent file.  Open in SliceViewer and select auto rebin and then change slices.
### Platforms affected
All

VickieLynch, 2017-04-14:
I tried to do an instrument view of TOPAZ with nightly and it never popped up the view. Tried mantidplot3.9 on the same machine at the same time and it popped up in a few seconds. Also LoadIsawPeaks for 6000 peaks now takes 5 minutes for TOPAZ peaks. Not sure these two problems are related to the original issue. If the problem is instrument 2.0, remember TOPAZ has approximately 1.5 million pixels.

SimonHeybrock, 2017-04-18:
@ianbush I think you refactored `LoadIsawPeak`. I remember you mentioned that it did slow down but the overall timings were well below the time for loading the instrument? Had you also tried this for TOPAZ?

SimonHeybrock, 2017-04-18:
@OwenArnold I built `master` at the point where we merged in the big "positions in `DetectorInfo` PR". At that point the instrument view still works fine. Since then, can you think of anything besides your changes for `ComponentInfo` that might have caused this?

OwenArnold, 2017-04-18:
@SimonHeybrock. Aside from the `ComponentInfo` changes, I'm not aware of anything that might have affected this.

ianbush, 2017-04-18:
@SimonHeybrock Yes, if I am remembering correctly that is right. The unit tests for `SCDCalibratePanels` (which was what I was really trying to refactor) showed some slow down but the effect in the tests for `SCDCalibratePanels` was small compared to the loading step. The slowdown for 6000 peaks may be traceable back to those changes, as the test may not deal with so many peaks.

SimonHeybrock, 2017-04-18:
@ianbush I do not remember any changes that would scale with the number of *peaks* (instrument size yes, but peak count?). Do you?

SimonHeybrock, 2017-04-18:
@OwenArnold I now checked also for the `ComponentInfo` and `IndexInfo` PR. Performance for the instrument view still seems to be the same.

SimonHeybrock, 2017-04-18:
@VickieLynch Regarding `BinMD`, it might be possible that this is already fixed by this PR: https://github.com/mantidproject/mantid/pull/19308. Could you check, or provide a file for testing?

ianbush, 2017-04-18:
@SimonHeybrock - you are correct, none of the changes I made should have any affect due to the number of peaks.

VickieLynch, 2017-04-18:
@SimonHeybrock Will test #19308 with large TOPAZ file

NickDraper, 2017-04-18:
Sorry for coming late to the discussion and somewhat jet lagged after my holiday

Is everyone confident they are looking at the same thing?  Would it be worth a bit of time @VickieLynch  creating a simple python script, or performance test that can show the difference that everyone can use?

OwenArnold, 2017-04-18:
@VickieLynch Can you try your tests with the changes from branch of PR #19395. This has greatly improved speed in some other areas.

ianbush, 2017-04-19:
The changes in #19395 were just merged into master - so @VickieLynch please try testing with that.

OwenArnold, 2017-04-19:
@ianbush @VickieLynch #19395 should improve things, but there is still a larger problem that I need to solve.

closed 2017-04-25 by peterfpeterson