mantidproject/mantid #19386: BASIS instrument view reverted to showing the neutronic positions

jmborr, 2017-04-14:
Instrument view in Mantid version 3.9.2 shows the real-space positions. A bug has been introduced since.

SimonHeybrock, 2017-04-18:
@jmborr After an initial look: https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument.cpp#L164

The changes in the PR you linked extract (i.e., remove) information from that parameter map. For example positions of detectors are now stored in the class `DetectorInfo`, instead of having relative positions changes in the parameter map.
However, the physical instrument would have a different detector, so the parameters for it should be untouched.

I will have to have a closer look, but first I need to track down some other issues, so this is just my initial thought.

SimonHeybrock, 2017-04-18:
I think I understand now what is happening, it seems indeed due to sharing of the parameter map. In particular, the problem is that the parameter map forwards to `DetectorInfo` when positions are read via the old legacy interface (such as `Detector::getPos()`). However, `DetectorInfo` of course contains positions for the neutronic instrument, and due to the shared parameter map with the physical instrument we get the wrong positions.

Will need to think a bit about a clean solution.

SimonHeybrock, 2017-04-18:
@jmborr There is a probably a quick fix that can be done in `Instrument`, but that will make the `InstrumentView` very slow. However, that "fix" does *not* work with the latest `BASIS` definition, due to changes done here: https://github.com/mantidproject/mantid/pull/19273. The changes there remove the tag `"indirect-neutronic-positions"`. Without that, Mantid does not create a physical instrument (see `InstrumentDefinitionParser`).

I will try and implement a proper fix (without `InstrumentView` slowdown) on Thursday (I am on leave tomorrow). Maybe you can fix the BASIS definition in the meantime?

jmborr, 2017-04-18:
@SimonHeybrock Thanks a million for taking the time and for spotting the bug in the latest instrument definition file. I'll fix the IDF right away

jmborr, 2017-04-20:
Pull request #19409 has addressed the first issue so now real space positions are shown when using an IDF containing both neutronic and real space positions.
However, the BASIS IDF in master has to be corrected to include real space positions as before issue #19258

jmborr, 2017-05-09:
Resolved _via_ #19421

closed 2017-05-09 by jmborr