mantidproject/mantid #19470: IndexingTest_SpectrumNumberTranslatorTest failing on Ubuntu 14.04

rosswhitfield, 2017-05-02:
`IndexingTest_SpectrumNumberTranslatorTest` has been failing on Ubuntu 14.04 for a while http://builds.mantidproject.org/job/master_clean-ubuntu/

```
test 179
    Start 179: IndexingTest_SpectrumNumberTranslatorTest

179: Test command: /usr/bin/cmake "-E" "chdir" "/home/builder/jenkins-linode/workspace/master_clean-ubuntu/build/bin/Testing" "/home/builder/jenkins-linode/workspace/master_clean-ubuntu/build/bin/IndexingTest" "SpectrumNumberTranslatorTest"
179: Test timeout computed to be: 300
179: Running 23 tests...
179: In SpectrumNumberTranslatorTest::test_spectrum_numbers_order_preserved:
179: /home/builder/jenkins-linode/workspace/master_clean-ubuntu/Framework/Indexing/test/SpectrumNumberTranslatorTest.h:116: Error: Test failed: Unknown error -1
179: /home/builder/jenkins-linode/workspace/master_clean-ubuntu/Framework/Indexing/test/SpectrumNumberTranslatorTest.h:117: Error: Test failed: Unknown error -1
179: /home/builder/jenkins-linode/workspace/master_clean-ubuntu/Framework/Indexing/test/SpectrumNumberTranslatorTest.h:118: Error: Test failed: Unknown error -1
179: /home/builder/jenkins-linode/workspace/master_clean-ubuntu/Framework/Indexing/test/SpectrumNumberTranslatorTest.h:119: Error: Test failed: Unknown error -1
179: ......terminate called after throwing an instance of 'std::system_error'
179:   what():  Unknown error -1
179: Child aborted
1/1 Test #179: IndexingTest_SpectrumNumberTranslatorTest ...***Failed    0.12 sec

0% tests passed, 1 tests failed out of 1
```

SimonHeybrock, 2017-05-02:
@rosswhitfield I suppose you do not know anything more specific? I just ran that test 10000 times without failure on Ubuntu 16.04.

rosswhitfield, 2017-05-02:
Nope, just that it seems to fails every time on Ubuntu 14.04.

SimonHeybrock, 2017-05-02:
Might be similar to this: https://github.com/google/benchmark/issues/67#issuecomment-66096431.

SimonHeybrock, 2017-05-03:
Related bug report for gcc: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60662, but this is not really giving a solution.
I can reproduce this in a 14.04 VM. Linking with `-lpthread` does not fix it though.

PR #19476 (2017-05-03): Add link flags for Ubuntu 14.04 required for std::call_once.
`std::call_once` uses `std::thread` internally. It segfaults on Ubuntu-14.04 unless special link flags are used:

http://stackoverflow.com/questions/8649828/what-is-the-correct-link-options-to-use-stdthread-in-gcc-under-linux#comment35723468_8649908

**To test:**

Code review should be sufficient (I tried this in a VM and it works, so I hope the nightly build does, too).

Fixes #19470.

No release notes.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testin [... cut]
closed 2017-05-03 by quantumsteve