mantidproject/mantid #19471: Python_CrystalFieldTest taking too long

OwenArnold, 2017-05-02:
[see test](https://github.com/mantidproject/mantid/blame/master/scripts/test/CrystalFieldTest.py) runs in ~30 seconds on my dev machine and the build servers. This is far too long for an automated unit test. This is hampering the development pipeline.

### Expected behavior

Runs in <= 1 seconds or gets made a system test

### Actual behavior

Sometimes runs > 30 seconds

### Steps to reproduce the behavior

### Platforms affected

All

mducle, 2017-05-02:
So, on my machine the test takes 18s. `import mantid` takes about 1s, so there's not really any way to make the test run faster than that. Otherwise what takes the time is `CrystalFieldFitTest` (the other tests in `CrystalFieldTest` and `BackgroundTest` take <0.1s).  

The test which takes the longest time is: `test_CrystalFieldFit_physical_properties` - which is my fault. I think we can move this to a system test or leave it out altogether. A previous test `test_api_CrystalField_physical_properties` already tests that these functions give correct numbers; the fit test is to check that it behaves sensibly when combined in a multi-dataset, multi-type (spectra and various physical properties) fit. On my machine it takes about 8s out of 17s total run-time.

Next the tests involving the monte-carlo parameter estimation engine take the most time around 4s for me. These are: `test_monte_carlo_single_spectrum`, `test_monte_carlo_multi_spectrum`, `test_estimate_parameters_cross_entropy` and `test_estimate_parameters_multiple_results`. They could be made faster by decreasing the number of sampling points, but this risks them failing to get the correct result.

Which leaves about 25% of the time (4s out of 17s) for the remaining tests. I don't think that we should put all of them into a system test. At the very least the tests in `CrystalFieldTest` and `BackgroundTest` should stay, and if we really want the test to run in less than 1s, then what can be done is just to check the syntax works and ignore any fit result, and setting the number of samples or iterations to 1, and maybe do the proper fitting in a system test.

OwenArnold, 2017-05-03:
@mducle - thanks for taking a look. I think a good first step would be to focus on the long running quick-wins you've outlined. Getting the test times down < 1 second is probably unrealistic - I'd be happy for this to be done in several stages if necessary.

mducle, 2017-05-05:
Since @mantid-roman is rewriting quite a lot of the Crystal Field back-end in issue #19341 these tests might change anyway, so I've just made some small changes to reduce the number of iterations / samples used (and also relaxed the convergence criteria) to make the tests run faster - I've got about a factor of 3 speed up.

PR #19488 (2017-05-08): Speed up CrystalField python tests
Changes the python `CrystalFieldTest` which uses `Fit` (or `EstimateParameters`) to reduce `MaxIterations` (or `NSamples`) used (with relaxed `assert` on the fitted parameters) so the tests runs faster. On my RHEL7 quad-core desktop the test used to take 18s, now it takes 5s. On the build servers the test now takes about 18s compared to around 30s before.

In the original issue #19471, it was suggested that these tests could all be moved into `SystemsTest` instead, which might be worthwhile for some of the longer running tests. However, these tests are not re [... cut]
closed 2017-05-08 by martyngigg