mantidproject/mantid #19549: BASIS IDF contains duplicate detector IDs

OwenArnold, 2017-05-15:
The BASIS [IDF](https://raw.githubusercontent.com/mantidproject/mantid/master/instrument/BASIS_Definition_20140101-.xml) tries to reuse detector ids

There should be no duplicates. `Instrument::markAsDetector` should not drop any Detectors. Much of this can be sanity checked by eye. 

1. ID Lists should contain unique detector id ranges.
1. ID Lists should be used only once within a given IDF.

This is blocking the rollout of Instrument 2.0 features.

jmborr, 2017-07-20:
@OwenArnold I have removed the duplicate `component` declarations for banks 1 through 4. I have verified that the new instrument definition file does not break reduction. How can I verify that loading the instrument does not try to reuse detector ID's?

OwenArnold, 2017-07-21:
@jmborr Thanks! and that's a good question.

If you delete the try-catch protection in here: https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument/InfoComponentVisitor.cpp#L99-L108, compile and run load empty instrument with the BASIS IDF(s) you should find that there are errors.

After these changes only #19552 IMAT, which @DTasev is working on, should be non-compliant.

jmborr, 2017-07-25:
@OwenArnold Can you review pull #20058?

closed 2017-07-27 by quantumsteve