mantidproject/mantid #19552: IMAT contains duplicate detector IDs

OwenArnold, 2017-05-15:
The IMAT [IDF](https://github.com/mantidproject/mantid/blob/master/instrument/IMAT_Definition_2015.xml) tries to reuse detector ids...this is not doing what the author expected. This affects **IMAT_Definition_2015.xml AND IMAT_Definition_2016.xml**. Both should be fixed as part of this.

For example, ID 1 is used twice because it appears once [here](https://github.com/mantidproject/mantid/blob/master/instrument/IMAT_Definition_2015.xml#L103), and then again in the range [here](https://github.com/mantidproject/mantid/blob/master/instrument/IMAT_Definition_2015.xml#L45). The second attempt leaves an parasitic detector - it is part of the full in-memory instrument but cannot actually be referenced since it's ID is invalid. See (`Instrument::markAsDetector` ) if you're interested in seeing how the detector ends up orphaned.

After fixing, much of this can be sanity checked by eye. 

1. ID Lists should contain unique detector id ranges.
1. ID Lists should be used only once within a given IDF.

This is blocking the rollout of Instrument 2.0 features.

DTasev, 2017-05-15:
So this does not affect the latest `IMAT_Definition.xml`? I will have a look, but need to consult Genoveva on what the structure of the instrument was then, I have no knowledge of how it looked.

Briefly looking, we want the monitors to be the first spectra, so would moving the detector IDs to start on the last monitor index + 2 (dead channels) be enough to solve this issue?

OwenArnold, 2017-05-15:
No, the latest one was not flagged up, so is probably ok. One simple way to check is to build in debug mode, run `LoadEmptyInstrument` on the relevant xml files, and check that in `Instrument::markAsDetector` you are not falling through on any of the detectors as you load.

DTasev, 2017-05-19:
I think I've solved the issue, but cannot test, as I cannot find where the old IDFs throw an error. Is it in:
- [Instrument.cpp L706](https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument.cpp#L706)
- [InfoComponentVisitor.cpp L69](https://github.com/mantidproject/mantid/blob/master/Framework/API/src/InfoComponentVisitor.cpp#L69) 

or somewhere else that I have missed?

OwenArnold, 2017-07-17:
@DTasev As far as I can tell your issue still remains in the 2015 (IDF) see my description and the links I have provided to sections in that IDF. 

To check any fixes.
1. Ensure that there are no duplicates of Ids. This can be easily done by eye.
2. To programatically check - Remove the try-catch parts of this https://github.com/mantidproject/mantid/blob/master/Framework/Geometry/src/Instrument/InfoComponentVisitor.cpp#L99-L108 so that it throws and then  run `LoadEmptyInstrument`

PR #20142 (2017-08-16): Fix IMAT IDFs from 2015 and 2016 causing issues due to overlapping IDs
Description of work.

The camera has been removed from the definition files. This should hopefully fix any ID collisions.

**To test:**

Fixes #19552. 

**Release Notes** 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)?
- [ ] Are the unit tests small and test the class [... cut]
closed 2017-08-16 by martyngigg