mantidproject/mantid #19586: FindPeaksMD does not find all peaks and finds wrong peaks

AntonPiccardoSelg, 2017-05-19:
Pascal reported an issue with the finding peaks and viewing them in the VSI. Displaying the peaks in the VSI can be improved by increasing the number of points and increasing the top percentile of points being displayed.

### Expected behavior

Should find all peaks

### Actual behavior

Does not find all peaks

### Steps to reproduce the behavior

1. Load the wish file 38416
2. Crop the loaded workspace with XMin=6000 to XMax=99000
3. Run ConvertToDiffractionMDWorkspace with QLab set
4. Check in the InstrumentView that you can see peaks
5. Run FindPeaksMD on the workspace and check in the instrument view that some peaks are not found and others are

AntonPiccardoSelg, 2017-05-19:
I contacted Vickie to check if she had encountered this issue and she replied that this is an issue in the US as well. They normally tweak the parameters to improve peak finding. Her reply was:
* PeakDistanceThreshold = 0.9 * 3.14 / max(cell length)
* DensityThresholdFactor = 100 
* Increase last factor by orders of magnitude to see if results improve

samueljackson92, 2017-05-19:
Pascal showed me this issue in person last week. I was surprised at how badly the algorithm performed. I'd advocate having a deeper exploration of why it is giving so many false positives/negatives. Is it just down to the  parameters used? Is there something else we could do to improve accuracy?

closed 2021-03-02 by stale[bot]