mantidproject/mantid #19641: Unscripted Testing Documentation

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Qt Assistant offline documentation
* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

AnthonyLim23, 2017-05-24:
None of the Algorithms,concepts, Interfaces or Fit Functions pages are showing up, just the missing page error. The first (index) page is fine for each one, just the links within the index do not work. 

An interesting feature is that the links to the algorithms works if they are selected via the release notes.

AnthonyLim23, 2017-05-30:
It all works as expected now.

closed 2017-05-30 by AnthonyLim23