mantidproject/mantid #19643: Unscripted Testing Mantidplot Group 4

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

PranavBahuguna, 2017-05-26:
* Filter Events Interface:
  - Due to the filtering algorithm not being performed asynchronously from the interface, a particularly taxing job will freeze Mantid until it is complete.
  - Giving a value of 0 for time interval in Filter By Time will cause the algorithm to run indefinitely. This appears to be a problem with the `GenerateEventsFilter` algorithm in general.

PranavBahuguna, 2017-05-26:
The OS environment this testing is occurring under is Windows.

closed 2017-06-01 by NickDraper