mantidproject/mantid #19647: Unscripted Testing Muon

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

As a minimum test http://www.mantidproject.org/Unscripted_Manual_Testing_MuonAnalysis.
For more info see http://www.mantidproject.org/Muon

AntonPiccardoSelg, 2017-05-24:
Questions from unscripted testing:
* Should "Show Error Bars" be unchecked?
* [This](http://docs.mantidproject.org/nightly/interfaces/MuonAnalysis_test_guides/Muon_Analysis_test_group_1_general.html#id3) didn't work for me. Why?
* [This](http://docs.mantidproject.org/nightly/interfaces/MuonAnalysis_test_guides/Muon_Analysis_test_group_1_general.html#id7) does not plot the fit in the data. Is this intentional?
* Graph is not being updated [here](http://docs.mantidproject.org/nightly/interfaces/MuonAnalysis_test_guides/Muon_Analysis_test_group_2_fitting.html#id3)

NickDraper, 2017-05-24:
@AnthonyLim23 Can you help out with these questions

AnthonyLim23, 2017-05-24:
I have had a chat with Anton. The document needs updating, I will work on it over the next few days.

AntonPiccardoSelg, 2017-05-24:
Other things:
* `Co-added` has the label enabled and there are two entries in the drop down menu instead of one.

AntonPiccardoSelg, 2017-05-25:
The interface functionality looks good. There are only some minor things which don't hold up sending this to beta testing.

closed 2017-05-25 by AntonPiccardoSelg