mantidproject/mantid #19649: Unscripted Testing ISIS SANS

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See http://www.mantidproject.org/SANS_Data_Analysis_at_ISIS

gemmaguest, 2017-05-30:
I'm testing this on Windows 7

gemmaguest, 2017-06-01:
I've finished testing and have raised several issues in #19772.

closed 2017-06-01 by gemmaguest