mantidproject/mantid #19653: Unscripted Testing Multi dataset fitting

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

AnthonyLim23, 2017-05-24:
I will do this once #19615 to ensure that the changes do not break anything else

AnthonyLim23, 2017-05-31:
There are a few quirks but they should not hold up the release. 

If a user sets fixs a set of parameters across multiple groups (e.g. A0 in fwd and A1 in bwd) then adds another function. Only the fixes for the currently displayed data are kept. 

If a user clicks the global box and then adds a function the global box is reset to false. 

There is a lack of documentation (particularly python)

closed 2017-06-01 by AnthonyLim23