mantidproject/mantid #19658: Unscripted Testing Vates

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

Including data loading in Vates via ParaView. For help see [[Testing_Notes_2.5.0|here]]

AntonPiccardoSelg, 2017-05-23:
# Tests

## Histogram Data

- [ ] Load data into VSI. Switch between different views
- [ ] Set preference to be multi-slice. Load data into VSI.
- [ ] Set various preferences for background colors and color scales.
- [ ] StandardView: load->slice->scale; delete filters
- [ ] StandardView: load->scale->slice; delete filters, but not in order
- [ ] MultisliceView: can create slices, shift them and open SliceViewer from slice
- [ ] ThreeSliceView: can shift slices via scroll button. 
- [ ] Change axes labels
- [ ] Change colorbar
- [ ] Change log scaling
- [ ] Play with rotation and centering buttons
- [ ] Load non-orthogonal data into the StandardView

## Event Data

- [ ]  Load data into VSI. Switch between different views
- [ ]  StandardView: load->slice->scale; delete filters
- [ ] StandardView: load->scale->slice; delete filters, but not in order
- [ ] MultisliceView: can create slices, shift them and open SliceViewer from slice
- [ ] ThreeSliceView: can shift slices via scroll button. 
- [ ] SplatterPlotView: change top percentile

## Event + Peak Data
- [ ]  Load Event data into SplatterPlotView. Drag and drop peak workspace
- [ ] Open peaks table; select peak from table (should zoom to it)

AntonPiccardoSelg, 2017-05-24:
Tested the above scenarios. It seems to function well. The only issue I have seen is already documented #18794

closed 2017-05-24 by AntonPiccardoSelg