mantidproject/mantid #19662: Unscripted Testing ORNL Powder Diffraction

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

See http://www.mantidproject.org/PowderDiffractionReduction

peterfpeterson, 2017-05-24:
First round of issues:
1. `SNSPowderReduction` works in the various modes, but is inefficient when chunking
    1. reloads full set of logs for each chunk
    1. re-reads calibration file when using `AlignAndFocusPowderFromFiles`
2. `AlignAndFocusPowder` does a bad job of cross checking input parameters (`ResampleX` vs `Params`)
3. `AlignAndFocusPowder` fails for out-of-place workspace. In-place is fine

peterfpeterson, 2017-06-05:
For 1.i, nothing is going to be done for this release because generating the event filter appears to be the slowest step.

closed 2017-06-05 by NickDraper