mantidproject/mantid #19664: Unscripted Testing ISIS Reflectometry

NickDraper, 2017-05-23:
1. Read http://www.mantidproject.org/Unscripted_Manual_Testing
1. Comment against this ticket the OS environment you are testing against.
1. Don't spend more than a few hours on the testing as fatigue will kick in.
1. If you find errors in the possibly scant documentation, then correct them.
1. Close the this issue once you are done.

If you find bugs:
* Search to see if an issue already exists
* create an issue if it is new
 * Urgent bugs or crashes should by against the current release, and assigned to a developer, then go and talk to the developer if possible.
 * Less urgent bugs should be against a subsequent release, and assigned if the correct developer is known.

### Specific Notes:

For testing procedures see http://www.mantidproject.org/Testing_ISIS_Reflectometry_GUI

DTasev, 2017-05-24:
The following issue was found:
- In the `Save ASCII` tab, if in the `List of Workspaces` a table workspace is selected, and in any format the `Save` button is clicked, Mantid will crash with a segfault. This might not be a bug in the GUI, but somewhere in the calls after.
- Fields in `Event Handling` tab allow input that's not numeric, is that intended?

PranavBahuguna, 2017-05-24:
Regarding the second issue, all alphabetical input is ignored in processing so this does not actually affect anything.

DTasev, 2017-05-28:
Closing as testing for release 3.10 was completed on Ubuntu 16 04

closed 2017-05-28 by DTasev