mantidproject/mantid #19669: MantidPlot crash when leaving a dialog open

gvardany, 2017-05-23:
MantidPlot crashes on close with segmentation fault when one leaves an auto-generated GUI of a runtime-registered custom algorithm open.

### Expected behaviour
MantidPlot should close normally.

### Actual behaviour
Segmentation fault on close

### Steps to reproduce the behaviour
1. Open MantidPlot.
2. From python script window create and register a custom algorithm, e.g.:
```
class SampleAlgorithm(PythonAlgorithm):
    def name(self):
        return 'SampleAlgorithm'
    def category(self):
        return 'Misc'
    def PyInit(self):
        pass
    def PyExec(self):
        pass
AlgorithmFactory.subscribe(SampleAlgorithm)
```
3. Open the auto-generated GUI of this new algorithm
4. Close MantidPlot while the algorithm dialog is open

### Platforms affected
Discovered on OS X, but probably ALL.

jmborr, 2017-05-23:
@martyngigg Would you take a look to this, please?

martyngigg, 2017-05-23:
@jmborr I'll take a look tomorrow. I think I understand the problem so hopefully it should't be too hard to fix.

PR #19676 (2017-05-24): Close all open algorithm dialogs manually before MantidPlot shutdown
Description of work.

Manually deletes all algorithm dialogs before shutting down MantidPlot rather than allowing to happen when Qt decides. We need to ensure that this is done in the case of Python algorithm dialogs that need to have access to an active Python interpreter.

**To test:**

Confirm crash before merging fix:

* start MantidPlot
* open a Python algorithm dialog
* close MantidPlot --> :boom: 

After merging fix try again and this time try a few different algorithm dialogs both C++ auto generated & custom dialogs a [... cut]
closed 2017-05-24 by rosswhitfield