mantidproject/mantid #19694: Indirect Data Analysis. Instrument selection not respected

DavidFair, 2017-05-25:
Copied from #19638 

Under S(Q, w) the current instrument is not respected. As this falls back to the user favourite instrument and that is probably correct I marked this issue as for release 3.11 (unless the fix is trivial enough for 3.10)

- Go into Help - First time setup
- Change the instrument (for example to ISIS/ALF). 
- Ensure Search Data Archive is enabled so you can see which file is loaded. 
- Go to the S(Q, w) tab, type in 21360 and hit Plot Input. 
- It will load the wrong data set (uses favourite instrument instead of instrument selected in the interface) and fail to plot.
- The instrument is not respected in the Moments tab either (see above)

SpencerHowells, 2017-05-25:
The interface is to be used only by instruments whose definition includes spectrometer = indirect inelastic. These are the only ones in the drop down menu. The instrument is changed using this and not via view for example. So far we have assumed that an user is not stupid enough to try it on a different type of instrument. However to be strictly correct we should check that it cannot be done.

SpencerHowells, 2017-05-25:
See also issue #14067

DavidFair, 2017-05-26:
The only confusing thing is it inconsistently works in the *Transmission* and *Symmetrise* tabs so there must be a fix there for this. 

Having seen this on other interfaces the fix is fairly trivial. Take the name of the instrument selected in the drop down on the interface, then prepend it to the run number string. Mantid will handle the rest from that point correctly.

SpencerHowells, 2017-06-05:
Not all the tabs use a raw file where run number can be input. 
_Transmission_ is a raw file - the Browse option is for *_raw.
_Symmetrise_ & _S(Q,w)_ require reduced files - the Browse option is for *_red.nxs files. The input is therefore NOT a run number but a file name eg iris26176_graphite002_red.nxs.
_Moments_ requires the output from S(Q,w) - the Browse option is for *_sqw.nxs.
All this is in the Help doc.

robertapplin, 2019-03-21:
The S(Q, W) takes a red file rather than raw. Could not reproduce this issue

closed 2019-03-21 by robertapplin