mantidproject/mantid #19718: Indirect Bayes - Closing interface while running causes crash

PranavBahuguna, 2017-05-26:
### Expected behavior

- In all three tabs of the interface, if the user closes the window before running has finished, it should either exit gracefully or they should be prevented from closing until running has finished.

### Actual behavior

- When user closes window while running, Mantid crashes.

### Steps to reproduce the behavior

- Open the Indirect Bayes interface and go to any of the three tabs (`ResNorm`, `Quasi`, `Stretch`).
- Enter files for Sample and Resolution (e.g. `irs26173_graphite002_red.nxs` and `irs26176_graphite002_red.nxs` respectively).
- Click `Run` then close the interface immediately - this should cause Mantid to crash.

### Platforms affected

Windows, likely others.

SpencerHowells, 2017-05-26:
Quasi & Stretch both use Fortran program and only run on Windows. Closing an interface will not stop the Fortran. The interaction with the Fortran may be there problem.

SpencerHowells, 2017-06-08:
Checked this out on Quasi. All the fitting works - log says BayesQuasi successful. It then crashes - I assume when it tries to return to the interface to deal with Plotting & Saving.

closed 2021-03-02 by stale[bot]