mantidproject/mantid #19727: Synchronization overhead in LoadNexusProcessed

quantumsteve, 2017-05-26:
### Expected behavior

Execution time in parallel loops should decrease with the number of core. 

### Actual behavior

Lots of time is spent waiting on a mutex in `EventWorksapceMRU::deleteIndex`. Is there anything more we can do about this?

[image]

### Steps to reproduce the behavior

I found this in #19726 by running LoadNexus on CORELLI data. @rosswhitfield @AndreiSavici Do we have a similar file in SystemTestData or StandardTestData?

### Platforms affected

All

quantumsteve, 2017-05-27:
Removing the locks [here](https://github.com/mantidproject/mantid/blob/b472472dcbe0a22a8fadc5cd18ba99fd5dddd9b2/Framework/DataObjects/src/EventWorkspaceMRU.cpp#L142) and [here](https://github.com/mantidproject/mantid/blob/b472472dcbe0a22a8fadc5cd18ba99fd5dddd9b2/Framework/DataObjects/src/EventWorkspaceMRU.cpp#L146) increases performance by a factor of two. There are a separate set of mutexes and locks in MRUList.h.

Since the number of threads is fixed at runtime, is such lock-free access to these vectors safe or easy made safe?

SimonHeybrock, 2017-05-29:
I might be possible, with at least one other change: ensuring that `data` is always allocated. I think if we manage to remove the `mutable` from `m_bufferedDataY` and `m_bufferedDataE` we should be fine?

SimonHeybrock, 2017-05-29:
Looking a bit closer, it might be more tricky: For the case of multiple algorithms accessing the same workspace, we need to ensure that we can run, e.g., `findY` and `deleteIndex` at the same time. In `findY` a pointer to the element is obtained (`MRUList::find`), and then a shared pointer is extracted. Currently the locks in `EventWorkspaceMRU` ensure that the element is not removed between the call to `MRUList::find` and copying the `shared_ptr` held by the obtained element.

closed 2017-06-20 by NickDraper