mantidproject/mantid #19796: Remove nexus python dependency

peterfpeterson, 2017-06-02:
[SANSUtility.py](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/SANSUtility.py) uses both `h5py` and the nexus python api (`nxs`). Change the calls to the latter to use the former (i.e. everything should use `h5py`). The bits that need to be redone are to remove lines 30-36, then redo the functions [`can_load_as_event_workspace`](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/SANSUtility.py#L1523) and [`get_measurement_time`](https://github.com/mantidproject/mantid/blob/master/scripts/SANS/SANSUtility.py#L1768).

**Note:** Make sure to branch off of `release-v3.10` so it can be merged into the release.

yxqd, 2017-06-02:
@AntonPiccardoSelg Could you point me to a test script that test this module so I can make sure my changes work? Thanks.

AntonPiccardoSelg, 2017-06-02:
@yxqd The tests for `get_measurement_time` function is indirectly covered via [this](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/analysis/SANSFileChecking.py). `can_load_as_event_workspace` does not seem to be covered. I will create a couple of test cases as a PR.

yxqd, 2017-06-04:
Thanks @AntonPiccardoSelg 

* get_measurement_time: $ cd /.../build; ./systemtest -R SANSFileChecking
* can_load_as_event_workspace: $ cd /.../build; ctest -R SANSUtility

PR #19801 (2017-06-06): replace nxs with h5py in SANSUtility
Fixes #19796 

Refs #19799 

**To test:**

Make sure following tests pass
* systemtest SANSFileChecking
* unittest SANSUtility

No release notes required.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/methods/functions?
- [ ] Are there unit/system tes [... cut]
closed 2017-06-06 by NickDraper