mantidproject/mantid #19816: AbinsModules to Python 3

AnthonyLim23, 2017-06-07:
Convert the AbinsModules to be compatible with python 3. These are group 15 in #18550 

The files are: 
`scripts/AbinsModules/AbinsConstants.py
scripts/AbinsModules/LoadCASTEP.py
scripts/AbinsModules/GeneralData.py
scripts/AbinsModules/AtomsData.py
scripts/AbinsModules/SData.py
scripts/AbinsModules/CalculateDWSingleCrystal.py
scripts/AbinsModules/CalculateS.py
scripts/AbinsModules/GeneralDFTProgram.py
scripts/AbinsModules/FrequencyPowderGenerator.py
scripts/AbinsModules/CalculatePowder.py
scripts/AbinsModules/LoadCRYSTAL.py
scripts/AbinsModules/InstrumentProducer.py
scripts/AbinsModules/AbinsData.py
scripts/AbinsModules/IOmodule.py
scripts/AbinsModules/DWSingleCrystalData.py
scripts/AbinsModules/KpointsData.py
scripts/AbinsModules/AbinsParameters.py
scripts/AbinsModules/Instruments/TwoDMap.py
scripts/AbinsModules/Instruments/ToscaInstrument.py
scripts/AbinsModules/Instruments/__init__.py
scripts/AbinsModules/Instruments/Instrument.py
scripts/AbinsModules/__init__.py
scripts/AbinsModules/PowderData.py
scripts/AbinsModules/AbinsTestHelpers.py
scripts/AbinsModules/SingleCrystalData.py
`scripts/AbinsModules/CalculateSingleCrystal.py`

dymkowsk, 2017-06-07:
Hi I just saw your issue. I am preparing a pull request which should solve this issue. It is currently in my branch  AbinsPython3.

AnthonyLim23, 2017-06-07:
:1:

dymkowsk, 2017-06-09:
I submitted a new pull request:
https://github.com/mantidproject/mantid/pull/19837

peterfpeterson, 2017-06-22:
This was done in #19837.

closed 2017-06-22 by peterfpeterson