mantidproject/mantid #19829: Python 3 Indirect Inelastic system test

louisemccann, 2017-06-08:
The system test for ISISIndirectInelastic is currently being skipped due to a syntax error. If this error is fixed the following tests fail:
```
ISISIndirectInelastic.IRISIqtAndIqtFit
ISISIndirectInelastic.IRISIqtAndIqtFitMulti
ISISIndirectInelastic.ISISIndirectInelasticBase
ISISIndirectInelastic.ISISIndirectInelasticCalibration
ISISIndirectInelastic.ISISIndirectInelasticConvFit
ISISIndirectInelastic.ISISIndirectInelasticDiagnostics
ISISIndirectInelastic.ISISIndirectInelasticElwinAndMSDFit
ISISIndirectInelastic.ISISIndirectInelasticIqtAndIqtFit
ISISIndirectInelastic.ISISIndirectInelasticIqtAndIqtFitMulti
ISISIndirectInelastic.ISISIndirectInelasticMoments
ISISIndirectInelastic.ISISIndirectInelasticReduction
ISISIndirectInelastic.ISISIndirectInelasticResolution
ISISIndirectInelastic.ISISIndirectInelasticTransmissionMonitor
ISISIndirectInelastic.OSIRISIqtAndIqtFit
ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti
```

louisemccann, 2017-06-08:
Also ISISIndirectSimulation and ISISIndirectLoadAscii

josephframsay, 2018-06-27:
All tests now pass on Python 2 and 3

closed 2018-06-27 by josephframsay