mantidproject/mantid #19831: Clean up code in MultipleFileProperty

martyngigg, 2017-06-08:
#19826 was merged in time for the release but comments on the code are valid and need to be addressed.

I think other clean ups would be worthwhile such as making some variables, `constexpr` rather than constructing them each time through the methods, e.g. [here](https://github.com/mantidproject/mantid/blob/0488b194c4a00d1b7046d689c7cd076ef9d85844/Framework/API/src/MultipleFileProperty.cpp#L247).

peterfpeterson, 2017-06-08:
It turns out you [can't use `constexpr` with `std::string`](https://stackoverflow.com/questions/27123306/is-it-possible-to-use-stdstring-in-a-constexpr) ... at least in c++11.

martyngigg, 2017-06-08:
Ok. We could just make them static local variables. C++11 did at least guarantee threadsafe initialization of such things.

PR #19836 (2017-06-09): Clean up code in MultipleFileProperty
Generic code cleanup/modernization.

**To test:**

Look things over and make sure builds pass.

Fixes #19831 and addresses comments from #19826.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/methods/functi [... cut]
closed 2017-06-09 by martyngigg