mantidproject/mantid #19850: When starting mantidplot from windows taskbar python fails

Anders-Markvardsen, 2017-06-14:
This bug was reported by Aidy Hillier yesterday and reproduced on my (windows) laptop.

### Steps to reproduce the behaviour
* Open Mantidplot
* Right click the Mantid icon on the bottom taskbar and select MantidPlot.exe

A window pops up saying "failed to initialize Python...." and the Result log message:

> Traceback (most recent call last):
  File "<string>", line 23, in <module>
  File "C:\MantidInstall\bin\mantidplot.py", line 10, in <module>
    import pymantidplot
  File "C:\MantidInstall\bin\pymantidplot\__init__.py", line 18, in <module>
    import pymantidplot.proxies as proxies
  File "C:\MantidInstall\bin\pymantidplot\proxies.py", line 15, in <module>
    import mantid
  File "C:\MantidInstall\bin\mantid\__init__.py", line 73, in <module>
    import mantid.kernel as kernel
  File "C:\MantidInstall\bin\mantid\kernel\__init__.py", line 14, in <module>
    from ._kernel import *
ImportError: DLL load failed: The specified module could not be found.
Previous initialization of Python failed, cannot retry.

### Platforms affected
Windows

DTasev, 2017-06-14:
I believe Mantid uses a script to append Mantid specific paths to the PYTHONPATH variable. The link shouldn't be to Mantidplot.exe, but the start `.bat` script. Was this the shortcut created by Mantid's installation or a custom one created by the user?

DTasev, 2017-06-14:
Potentially look into Qt wrapper or direct calls to WinAPI for adding tasks to win7/win10 right click on taskbar.

closed 2021-03-01 by stale[bot]