mantidproject/mantid #19864: Seperate out slow system tests from PR builds

NickDraper, 2017-06-15:
Slack Channel

Branch
https://github.com/mantidproject/mantid/tree/Dev2017_flag_and_skip_slow_tests

DavidFair, 2017-06-15:
Added a check in the system test script ran by Jenkins: 
If the name does not contain *pull_requests* (i.e. we are the master or release build) we run the slow system tests, otherwise we skip them.

NickDraper, 2017-06-15:
Potential slow tests identified, marked on spreadsheet, these have not been labelled in the system test code yet.

NickDraper, 2017-06-15:
Added a flag for enabling slow system tests to be run. Slow tests are not run by default. A boolean parameter isSlow has been added to the stressting.MantidStressTest to mark where a test is slow.

NickDraper, 2017-06-15:
change to "--exclude-in-pull-requests"

NickDraper, 2017-06-15:
will need to return specific exit code, see eg __verifyMemory in stresstesting.py

louisemccann, 2017-06-16:
@DavidFair you need to re-do your build script changes

DavidFair, 2017-06-19:
I'm going to revert my changes - talking to @peterfpeterson we can pass extra args (i.e. `--exclude-in-pull-requests`) on Jenkins which is the preferred method.

DTasev, 2017-06-19:
When the final solution is decided on, I'm happy to change the top X slow algorithms to have the `slow` flag (or whichever name we set upon - `excludeInPullRequests`, etc..)

PR #19910 (2017-06-22): Dev2017 flag and skip slow tests
A function `excludeInPullRequests` has been added to the base class for system tests, allowing for especially slow or unnecessary tests to be skipped for every PR build and only run nightly. Tests will be excluded if this function returns true.

When calling `systemtest` from the command line, all tests will be run by default. A flag `--exclude-in-pull-requests` has been added to exclude any systems test with the above function returning true.

A few of the slowest system tests have been updated to be skipped, see https://docs.google.com/spreadsheets/d/1uzAhnFM [... cut]
closed 2017-06-22 by martyngigg