mantidproject/mantid #19890: Investigate clcache for speeding up Windows builds

gemmaguest, 2017-06-19:
We did some preliminary work on this at the Mantid Developers' Workshop. This issue is to record our progress.

https://github.com/frerich/clcache
The below was tested on Windows 10

### Prerequisites (pyinstaller and python versions)
- clcache.py requires Python 3 (we used 3.5)
- The installation instructions below use pyinstaller. This only works with python versions 3.3 to 3.5 (and 2.7, but clcache doesn't work with Python 2)
- Make sure you have the correct version of python as the default version in your path before installing pyinstaller.
- Install pyinstaller: `pip install pyinstaller`
- If you change your version of python but already had pyinstaller, you may need to reinstall it to get it to recognise the new version.

### Installing clcache
There are several ways of installing clcache. The one we had the most success with was as follows. (It would be easier to install via Choco but it did not seem to be available.)
- Clone the repository from https://github.com/frerich/clcache
- cd into the repository
- Run `pyinstaller -F clcache.py` (the -F avoids a problem we had where the python dll was not found)
- clcache.exe is then available in the dist/ subdirectory
- You can delete the build and dist subdirectories if you need to do a clean build of clcache.

### Integration for Visual Studio
This was the only integration method we managed to get working from the instructions here: https://github.com/frerich/clcache/wiki/Integration
- Find the directory where the cl.exe used by Visual Studio resides. There may be several versions of cl.exe, so you may need to run a build and check the process that is running in Task Manager to find the correct one. For us, this was `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64_x86\`
- Rename cl.exe to e.g. cl_original.exe
- Rename cl.exe.config to e.g. cl_original.exe.config
- Copy the generated clcache.exe file to cl.exe
- Set CLCACHE_CL environment variable to point to cl_original.exe (make sure you don't set it to cl.exe by mistake! It must point to the renamed original.)

### Running the build
- Run `cl.exe -s` to see statistics
- Start a build and rerun `cl.exe -s` to see the changes. You should see the cache size increase.
- Clean and rebuild and you should see some cache hits. Unfortunately there are known problems with a clean rebuild documented here: https://github.com/frerich/clcache/wiki/Caveats. We tried the suggested fix of setting TrackFileAccess in a particular .vcxproj file to false but this didn't seem to work. 

### What next
- Check if we get cache hits after switching branches
- Try editing all .vcxproj files (we only tried for one specific module)
- Test on Windows 7

gemmaguest, 2017-06-19:
Hi @ianbush, I've added you to this issue but feel free to unassign yourself if you don't plan to do any more on it. I won't have time for a few weeks but I'll probably take another look at some point.

ianbush, 2017-06-20:
Hi @gemmaguest - I will take a look at this if I get a chance again, a lot of it is just starting things in the background and coming back to see what happened. I will add a couple of other notes to the ticket.

gemmaguest, 2019-07-25:
Closing because there don't seem to be any suitable options at the moment

closed 2019-07-25 by gemmaguest