mantidproject/mantid #19901: Symbol visibililty in Parallel sub-project

peterfpeterson, 2017-06-20:
The [merge](https://github.com/mantidproject/mantid/commit/d23449f3972960a0fddb121edb8be2816d701355) of #19280 introduced a linker issue in gcc 6.3.1 (fedora 25, cmake 3.8.0) when `CMAKE_CXX_VISIBILITY` is set to `hidden`. Changing it to `default` removes the issue. From the linker error messages, it appears to be specific to `StorageMode` in `MantidParallel/StorageMode.h`.

```
Linking CXX executable bin/PythonInterfaceCppTest
FAILED: bin/PythonInterfaceCppTest
: && ccache /usr/lib64/ccache/c++  -fopenmp -Wsuggest-override -Wall -Wextra -Wconversion -Winit-self -Wpointer-arith -Wcast-qual -Wcast-align -fno-common -Wno-deprecated -Wno-write-strings -Wno-unused-result -pedantic -Woverloaded-virtual -fno-operator-names -O2 -DNDEBUG   Framework/PythonInterface/test/cpp/CMakeFiles/PythonInterfaceCppTest.dir/RunPythonScriptTest.cpp.o Framework/PythonInterface/test/cpp/CMakeFiles/PythonInterfaceCppTest.dir/PySequenceToVectorTest.cpp.o Framework/PythonInterface/test/cpp/CMakeFiles/PythonInterfaceCppTest.dir/PythonObjectInstantiatorTest.cpp.o Framework/PythonInterface/test/cpp/CMakeFiles/PythonInterfaceCppTest.dir/PropertyWithValueFactoryTest.cpp.o Framework/PythonInterface/test/cpp/CMakeFiles/PythonInterfaceCppTest.dir/PythonInterfaceCppTest_runner.cpp.o  -o bin/PythonInterfaceCppTest  -Wl,-rpath,/home/pf9/build/mantid/bin:/home/pf9/build/mantid/bin/mantid/kernel:/home/pf9/build/mantid/bin/mantid/api bin/libMantidAPI.so bin/libMantidGeometry.so bin/libMantidKernel.so bin/mantid/kernel/_kernel.so bin/mantid/api/_api.so /usr/lib64/libboost_python.so /usr/lib64/libboost_date_time.so /usr/lib64/libboost_regex.so /usr/lib64/libboost_serialization.so /usr/lib64/libPocoFoundation.so /usr/lib64/libPocoUtil.so /usr/lib64/libPocoXML.so /usr/lib64/libPocoNet.so /usr/lib64/libPocoCrypto.so /usr/lib64/libPocoNetSSL.so /usr/lib64/libpython2.7.so /usr/lib64/libNeXus.so /usr/lib64/libNeXusCPP.so /usr/lib64/libtbb.so /usr/lib64/libtbbmalloc.so -Wl,-rpath-link,/home/pf9/build/mantid/bin && :
bin/libMantidAPI.so: undefined reference to `Mantid::API::Workspace::storageMode() const'
collect2: error: ld returned 1 exit status
```
This *should* be an issue with windows as well, but isn't. Assuming that this will get fixed as tools advance, this probably doesn't warrant a large effort to fix.

Moderately related is that `MantidParallel/DllConfig.h` should have the attached file applied to it to make it match all of the similar export files.
[patch.txt](https://github.com/mantidproject/mantid/files/1089587/patch.txt)

SimonHeybrock, 2017-06-21:
@peterfpeterson Having read the discussion on Slack about this yesterday, I am not sure what the issue is? Does the patch you provided fix the issue or not?

peterfpeterson, 2017-06-21:
Changing `CMAKE_CXX_VISIBILITY` to `default` is the only way I've made things work. The patch is just to make `DllConfig.h` match all the others in the repository. The issue is mostly to capture the conversation.

closed 2021-03-03 by stale[bot]