mantidproject/mantid #19918: segmentation fault running AlgorithmMPITest

quantumsteve, 2017-06-22:
### Expected behavior

Mantid builds and runs all tests with open source clang
### Actual behavior

segmentation fault running AlgorithmMPITest
http://builds.mantidproject.org/job/master_clean-osx-10.10-openmp/259/console

I tried running it through lldb, but the backtrace wasn't terribly helpful.

```
.FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlgAnyModeToParallel::StorageMode::Distributed-[Notice] FakeAlgAnyModeToParallel::StorageMode::Distributed started
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
FakeAlg0To1Parallel::StorageMode::MasterOnly-[Notice] FakeAlg0To1Parallel::StorageMode::MasterOnly successful, Duration 0.00 seconds
[... 52 lines cut ...]
(lldb) bt
* thread #1: tid = 0x13c96d1, 0x00007fff8b889286 libsystem_kernel.dylib`__pthread_kill + 10, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff8b889286 libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff930a89f9 libsystem_pthread.dylib`pthread_kill + 90
    frame #2: 0x00007fff8d50b9a3 libsystem_c.dylib`abort + 129
    frame #3: 0x00007fff881471cb libsystem_malloc.dylib`free + 428
    frame #4: 0x0000000102437a70 libomp.dylib`__kmp_free_implicit_task + 30
    frame #5: 0x000000010242fc54 libomp.dylib`__kmp_reap_thread(kmp_info*, int) + 126
    frame #6: 0x000000010242ea80 libomp.dylib`__kmp_internal_end_library + 439
    frame #7: 0x00007fff5fc13010 dyld`ImageLoaderMachO::doTermination(ImageLoader::LinkContext const&) + 212
    frame #8: 0x00007fff5fc022e2 dyld`dyld::runAllStaticTerminators(void*) + 67
    frame #9: 0x00007fff8d50c71d libsystem_c.dylib`__cxa_finalize_ranges + 322
    frame #10: 0x00007fff8d50ca20 libsystem_c.dylib`exit + 55
    frame #11: 0x00007fff8a82d5d0 libdyld.dylib`start + 8
```

### Steps to reproduce the behavior

1. `CC=/usr/local/opt/llvm/bin/clang CXX=/usr/local/opt/llvm/bin/clang++ cmake -G Ninja <source_dir>`
2. `ninja && ninja AllTests`
3. `ctest -R APITest_AlgorithmMPITest`

### Platforms affected

MacOS with llvm 4.0

quantumsteve, 2017-06-22:
The failures are somewhere in `test1To1StorageModeTransition`

SimonHeybrock, 2017-06-23:
Tried running for a long time on Linux/gcc, no failures.

Trying to build with clang on Linux now, getting linker errors for things in the standard library and boost? Is there anything special I need to do?

Also, by default clang does not seem to support OpenMP, but from your issue description I understand this is an OpenMP-enabled build?

quantumsteve, 2017-06-23:
@SimonHeybrock Yes, On MacOS homebrew always builds the llvm openmp library. On Ubuntu you need to also install `libomp5` and `libomp-dev`. I'm not sure about other distributions.

quantumsteve, 2017-06-23:
I'm seeing similar failures on Ubuntu 17.04 and clang
[AlgorithmMPITest.txt](https://github.com/mantidproject/mantid/files/1098679/AlgorithmMPITest.txt)

quantumsteve, 2017-06-23:
@rosswhitfield added the OpenMP to the ArchLinux buildserver. Might make it easier to track down this issue. http://builds.mantidproject.org/job/master_clean-archlinux-clang/

SimonHeybrock, 2017-06-26:
@quantumsteve Are there any other clang specifics I need to install? CMake fails with some missing modules, such as `boost-regex` and `MuParser`?

quantumsteve, 2017-06-26:
There are issues with the newer GCC ABI and older versions of clang. The simplest solution is to use clang v3.9 or later.

https://bugs.llvm.org/show_bug.cgi?id=23529

SimonHeybrock, 2017-06-26:
Using 3.9 I still get weird linker errors with libstdc++. A quick test with libc++ also fails (differently).

Don't we have a guide how to build with clang on Linux?

quantumsteve, 2017-06-26:
@SimonHeybrock Which distribution are you using? I can't reproduce your issue on Ubuntu 17.04 or our ArchLinux build server.

SimonHeybrock, 2017-06-27:
@quantumsteve Ubuntu 16.04.

I now tried setting up a 17.04 VM but I am getting a lot of dependency issues. There appears to be no dev package / working ppa yet?

SimonHeybrock, 2017-06-27:
@quantumsteve @OwenArnold tried this with Apple Clang (`Apple LLVM version 7.0.2 (clang-700.1.81)`, not 4.0) and does not see the issue. Maybe you can do some more debugging to figure out what is going on?

quantumsteve, 2017-09-21:
I upgraded to clang 5.0 and unfortunately this intermittent issue remains.

quantumsteve, 2017-12-11:
I'm seeing a similar strange segfault with the open source clang running`AlgorithmsTest_RebinTest`

Here's the backtrace

```
$ lldb /Users/svh/Documents/MantidProject/build/bin/AlgorithmsTest -- RebinTest
(lldb) target create "/Users/svh/Documents/MantidProject/build/bin/AlgorithmsTest"
Current executable set to '/Users/svh/Documents/MantidProject/build/bin/AlgorithmsTest' (x86_64).
(lldb) settings set -- target.run-args  "RebinTest"
(lldb) run
Process 79514 launched: '/Users/svh/Documents/MantidProject/build/bin/AlgorithmsTest' (x86_64)
Running 25 testsRebin-[Notice] Rebin started
PropertyManager-[Error] Property "PARAMS" is not set to a valid value: "Enter values for this property".
PropertyManager-[Error] Property "PARAMS" is not set to a valid value: "Enter values for this property".
Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
PropertyManager-[Error] Property "PARAMS" is not set to a valid value: "Enter values for this property".
PropertyManager-[Error] Property "PARAMS" is not set to a valid value: "Enter values for this property".
Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.ConvertToPointData-[Notice] ConvertToPointData started
ConvertToPointData-[Notice] ConvertToPointData successful, Duration 0.00 seconds
Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.MaskBins-[Notice] MaskBins started
MaskBins-[Notice] MaskBins successful, Duration 0.00 seconds
Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
MaskBins-[Notice] MaskBins started
MaskBins-[Notice] MaskBins successful, Duration 0.00 seconds
Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
.Rebin-[Notice] Rebin started
Rebin-[Notice] Rebin successful, Duration 0.00 seconds
....Rebin-[Error] Error in execution of algorithm Rebin:
Rebin-[Error] MatrixWorkspace: Parallel support for XMin and XMax not implemented.
Rebin-[Warning] Error in execution of algorithm Rebin:
[... 44 lines cut ...]
Rebin-[Warning] MatrixWorkspace: Parallel support for XMin and XMax not implemented.
.OK!
Process 79514 stopped
* thread #1: tid = 0xa9bfbf, 0x0000000103e4ba59 libomp.dylib`__kmp_dephash_free_entries + 44, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x0000000103e4ba59 libomp.dylib`__kmp_dephash_free_entries + 44
libomp.dylib`__kmp_dephash_free_entries:
->  0x103e4ba59 <+44>: movq   (%rcx,%r12,8), %r14
    0x103e4ba5d <+48>: testq  %r14, %r14
    0x103e4ba60 <+51>: je     0x103e4bae7               ; <+186>
    0x103e4ba66 <+57>: movq   %r14, %rbx
(lldb) bt
* thread #1: tid = 0xa9bfbf, 0x0000000103e4ba59 libomp.dylib`__kmp_dephash_free_entries + 44, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
  * frame #0: 0x0000000103e4ba59 libomp.dylib`__kmp_dephash_free_entries + 44
    frame #1: 0x0000000103e4bba0 libomp.dylib`__kmp_dephash_free + 18
    frame #2: 0x0000000103e2b590 libomp.dylib`__kmp_free_implicit_task + 30
    frame #3: 0x0000000103e240d6 libomp.dylib`__kmp_reap_thread(kmp_info*, int) + 123
    frame #4: 0x0000000103e22efe libomp.dylib`__kmp_internal_end_library + 437
    frame #5: 0x00007fff5fc133bc dyld`ImageLoaderMachO::doTermination(ImageLoader::LinkContext const&) + 212
    frame #6: 0x00007fff5fc0218b dyld`dyld::runAllStaticTerminators(void*) + 67
    frame #7: 0x00007fff8ac5e463 libsystem_c.dylib`__cxa_finalize_ranges + 345
    frame #8: 0x00007fff8ac5e767 libsystem_c.dylib`exit + 55
    frame #9: 0x00007fff9aa175b4 libdyld.dylib`start + 8
    frame #10: 0x00007fff9aa175ad libdyld.dylib`start + 1
(lldb) 
```

quantumsteve, 2018-05-02:
Similar (same) error present in ubuntu 18.04 with `CC=clang CXX=clang++ cmake -G Ninja SRC_PATH`

closed 2021-03-02 by stale[bot]