mantidproject/mantid #19962: Inconsistent use of number density

ianbush, 2017-07-05:
In `CalculateSampleTransmission` the mass density can be used. This uses the material builder to calculate a number density. The number density calculated by the material builder is in molecules/Angstrom^3, not atoms/Angstrom^3.

The same mistake may exist elsewhere, this should be checked.

peterfpeterson, 2017-07-06:
One of the places that needs to be checked is [here](https://github.com/mantidproject/mantid/pull/12758/files#diff-7e6895aa1590493fe92220f55c3d630bR262)

peterfpeterson, 2017-07-06:
Also, this would be a good time to add a "packing fraction" (set/get) and "effective density" (set only) to the `Material`/`Sample` object(s).

ianbush, 2017-07-07:
@peterfpeterson I agree those things would be good to add, but I would rather keep this PR to the bug/docs fixes. Those changes would be good in a PR following on from this.

PR #19978 (2017-08-04): Inconsistent use of number density
The main change is to ensure `MaterialBuilder` uses atoms per cubic Angstrom for the number density, instead of formula units per cubic Angstrom.

There are some documentation changes to ensure that it is clear that `atoms / Angstrom^3` are used for number density and not anything else.

**To test:**

Check unit tests pass. Is there anywhere else the documentation should be clarified? Is there any knock-on effect on user scripts we should worry about, or is this just a bug fix?

Fixes #19962

---

#### Reviewer ####

Please comment on the following ([full des [... cut]
closed 2017-08-04 by martyngigg