mantidproject/mantid #19964: Cleanup related to recent Instrument-2.0 changes

SimonHeybrock, 2017-07-05:
- move `DetectorInfo` init into `InfoComponentVisitor`
- safe init proceedure such that `DetectorInfo` will be guaranteed to have a valid `ComponentInfo` and vice versa. Could be done by making `InfoComponentVisitor` a factory for both (and forbidding other ways of instantiation -- would make testing hard?)
- remove `relativePosition` and `relativeRotation`?

SimonHeybrock, 2017-08-01:
Regarding 6.):
We cannot remove this directly since `SpectrumInfo::detector()` and `DetectorInfo::detector()` return a reference, i.e., we must store the `shared_ptr<Detector>` somewhere. Probably the way to solve this would be to return a pointer instead. Currently the `detector()` method is still used a lot (mainly to get the detector ID), so we should first reduce the number of uses.

SimonHeybrock, 2018-09-10:
@OwenArnold Most items look outdated, but some are maybe still relevant...? Should I keep this open, or close?

OwenArnold, 2018-09-10:
@SimonHeybrock yes, some items done, but would like to keep this open for future reference and in case we get Bhuvan back next year (that kind of thing).

closed 2021-03-02 by stale[bot]