mantidproject/mantid #20000: Rebin fails on Exercise 4 of Introduction to Python in Mantid (Extending Python With Mantid)

brandonhewer, 2017-07-13:
The current automated solution - that I've written. Rebin seems to fail when monitors are included - succeeds by excluding monitors. Tested manually and achieved same outcomes.

``` python
from mantid.kernel import *
from mantid.api import *
from mantid.simpleapi import Load, ConvertUnits, Rebin
from numpy import *

class ConvertToEnergy(PythonAlgorithm):

    def PyInit(self):
        self.declareProperty(FileProperty(name="Filename",
                                          defaultValue="",
                                          action=FileAction.Load))
        self.declareProperty(name="Ei", defaultValue=0.0,
                             validator=FloatMandatoryValidator())
        self.declareProperty(FloatArrayProperty(name="BinParams",
                                                validator=FloatArrayLengthValidator(3),
                                                direction=Direction.Input))
        self.declareProperty(WorkspaceProperty(name="OutputWorkspace",
                                               defaultValue="",
                                               direction=Direction.Output))

    def PyExec(self):
        _tmpws = Load(Filename=self.getPropertyValue("Filename")) # Excluding Monitors fixes this
        _tmpws = ConvertUnits(InputWorkspace=_tmpws, Target="DeltaE",
                              EMode="Direct", EFixed=self.getPropertyValue("Ei"))
        _tmpws = Rebin(InputWorkspace=_tmpws, Params=self.getPropertyValue("BinParams"))
        output = WorkspaceFactory.create(_tmpws, NVectors=1)
        output.setX(0, _tmpws.dataX(0))
        spectraSum = np.zeros(_tmpws.getNumberBins())

        for i in range(_tmpws.getNumberBins()):
            sum = 0

            for j in range(_tmpws.getNumberHistograms()):
                sum += _tmpws.dataY(j)[i]

            spectraSum[i] = sum

        _tmpws.delete()
        output.setY(0, spectraSum)
        self.setProperty("OutputWorkspace", OutputWorkspace)

AlgorithmFactory.subscribe(ConvertToEnergy)
```

DavidFair, 2017-07-25:
@brandonhewer Which file were you trying to load with this algorithm?

josephframsay, 2017-07-25:
```
from mantid.kernel import *
from mantid.api import *
from mantid.simpleapi import Load, ConvertUnits, Rebin
from numpy import *

_tmpws = LoadRaw(Filename="MAR11015.raw", LoadMonitors="Include") # Excluding Monitors fixes this
_tmpws = ConvertUnits(InputWorkspace=_tmpws, Target="DeltaE",
                      EMode="Direct", EFixed="12.9729")
_tmpws = Rebin(InputWorkspace=_tmpws, Params="-11,0.01,11")
```

PR #20109 (2017-08-02): Improved error message in Rebin when sentinel values not set
**Description of work.**

It was found that, when rebinning a workspace containing monitors after running ConvertUnits on that workspace, an InvalidBinEdgesError was thrown. This is due to cells in the monitor being assigned to `DBL_MIN` as no sensible value could be produced. No more elegant fix could be found, but a more meaningful error message is now given when the above occurs.

**To test:**
```
from mantid.kernel import *
from mantid.api import *
from mantid.simpleapi import Load, ConvertUnits, Rebin
from numpy import *

_tmpws [... cut]
closed 2017-08-02 by SimonHeybrock